thonny/thonny #941: Update French translation

NathanBnm, 2019-10-13:
Hi! I don't really know how you proceed so I prefer to open an issue here.

I just completed the missing French translation and fixed some strings.

It would be great if you can implement this new translation in the next version of Thonny!

I noticed that there were strings in the app that are not in the translation. Can you explain how to add them in order to translate them too?

aivarannamaa, 2019-10-13:
Thanks, I'll update French translations form Poeditor before next release.

> I noticed that there were strings in the app that are not in the translation. 

Some strings are left untranslated on purpose. Which strings do you mean?

NathanBnm, 2019-10-15:
Great!

I'm thinking about the "Device" menu with the "Where are all the commands?" submenu and the associated dialog.

There is also the "<untitled>" of a new file.

There are also the subcolumns in the panels like "Stack" and others. Column names are like "Name", "Value" or "Function".

There is the string "If last command raised an exception then this view will show the stacktrace." in the Exception panel too.

Let me know if you have any questions.

aivarannamaa, 2019-10-30:
I'm sorry about the delay.

> I'm thinking about the "Device" menu with the "Where are all the commands?" submenu and the associated dialog.

This is temporary, so I don't want to make this translatable.

> There is also the "" of a new file.

What do you mean?

> There are also the subcolumns in the panels like "Stack" and others. Column names are like "Name", "Value" or "Function".
> There is the string "If last command raised an exception then this view will show the stacktrace." in the Exception panel too.
> And finally there is the "Save On Close" dialog.

Thanks! I'll include these in version 4.0. I don't want to disturb translations anymore for 3.2 line, which will probably get only one bug-fix release.

NathanBnm, 2019-10-30:
>What do you mean?
>
Whoops I forgot to film the string. I think it's the string displayed when creating a new file. Something like "untitled" or "unnamed".

closed 2021-12-25 by aivarannamaa