thonny/thonny #944: raspberry pi4 thonny steals modules from terminal

irreverentperson, 2019-10-14:
after using thonny, some packages are no longer available in the terminal but they work in thonny.

aivarannamaa, 2019-10-14:
Can you be more specific -- which modules? Are you using system provided Thonny? Did you change anything in "Tools => Options => Interpreter"? Are you not confusing commands `python` and `python3`?

irreverentperson, 2019-10-14:
Board, busio, adafruit-gps modules. They all worked in terminal before I installed them in thonny  through its package installer. I have tried python, python3, sudo python and sudo python3. It is system provided thonny. Only thing I did was install the packages on thonny. Now I get errors in the terminal for those modules

aivarannamaa, 2019-10-14:
> They all worked in terminal before I installed them in thonny through its package installer.

Why did you install them in Thonny if you already had them installed? Did you use a virtual environment?

> Now I get errors in the terminal for those modules

What does the error messages say?

irreverentperson, 2019-10-14:
I installed them in thonny because I was getting import error no module
named ... now I get that error when I run the same code in terminal.

aivarannamaa, 2019-10-14:
What do you see when you go to "Tools => Options => Interpreter"?

Did you open terminal from Thonny (Tools => Open system shell) or from System menu?

Which Thonny version are you using?

irreverentperson, 2019-10-14:
Tools options interpreter says

 "the same interpreter which runs thonny.
Details
 /usr/bin/python3"

I open thonny by saving a .py from the terminal then I click on the file
and it automatically opens thonny.

It is thonny.desktop @29:1

aivarannamaa, 2019-10-15:
In recent Thonny versions, "Manage packages" dialog displays installation location for each package. Where are the problematic modules installed?

irreverentperson, 2019-10-15:
Sorry, I wiped the pi and started over. Removed thonny.

closed 2019-10-23 by aivarannamaa