thonny/thonny #945: RDS Server Thonny Install ( One only user can use it at the same time on the server  )

gclay908, 2019-10-15:
We have an RDS Server with Python with Thonny installed, but when we have multiple people using the server at one time, only one of the users can use Thonny at one time. 

Is there away for everyone to be able to use this at one time on the same server ?

aivarannamaa, 2019-10-15:
Which version are you using?

gclay908, 2019-10-16:
3.1

Ill see if the option in 3.2 fixes anything.

gclay908, 2019-10-16:
I have check 3.2 and does work sort of. 

I changed the option as an admin and then tried to run it as a different user ( At the same time ) and worked, but everytime a new user opens thonny, it still has the option ticked ( Only run 1 instance ) and then if someone else tried to get on after the user it doesnt open still. 

Have you got a script we can run to disable that option everytime they run thonny ? 

Thanks :)

aivarannamaa, 2019-10-16:
In 3.2.1 Thonny is supposed to run in RDS even with single instance mode turned on (I haven't tried it myself though). Did you try it?

gclay908, 2019-10-16:
Yeah still only allowed 1 instance to run

aivarannamaa, 2019-10-16:
You can prepare initial configuration file for new users: https://github.com/thonny/thonny/wiki/DeploymentOptions#pre-configuring-an-all-users-installation

Currently user configuration is stored under roaming appdata and single instance coordination file is under local appdata. How are these set up in RDS? Would it help if single instance coordination also takes place under roaming appdata?

aivarannamaa, 2019-10-16:
In other words, in order to make single instance mode work in multiuser systems, I need a location which is accessible to current user and not shared with other users.

gclay908, 2019-10-16:
Hi,

Local users %AppData% would do it. Students don’t have roaming profiles.

Also while asking can Thonny use another preconfigured Python/Pygame instance as currently it has its own so when staff ask for additional applications/environments we have to add them to both Python environments.

If we know where the App is expecting to read its config file we can pre populate that for the user at login. Items such as default save path which should be H:\ can also be set as current Thonny sticks files in the user profile on the local RDS server?

Is this ok ?

Thanks
George

aivarannamaa, 2019-10-16:
Yes, you can prescribe different interpreter for running programs. You can do it manually from Tools => Options => Interpreter. After changing this and closing Thonny, you can see the result in `%APPDATA%\Thonny\configuration.ini`. The interesting attributes are `backend_name` in `[run]` section and `path` in `[CustomInterpreter]` section.

You can prescribe initial working directory by assigning it to `working_directory` in the `[run]` section. Current directory is directly tied to the directory shown in Fiels view (View => Files) and it will be initial directory in file dialogs.

If is also possible to change the location of the configuration file: https://github.com/thonny/thonny/wiki/DeploymentOptions#location-of-thonnys-user-data-configuration-and-log-files

closed 2019-10-18 by gclay908