thonny/thonny #946: REPL doesn't work with ESP8266 dev board

JennaSys, 2019-10-15:
Trying to use a CP2104 based "download" board like [this one](https://www.aliexpress.com/item/33062578172.html) and the REPL gives an error after establishing a connection:
```
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\backend.py", line 257, in _interrupt_to_raw_prompt
    raise TimeoutError("Can't get to raw prompt. Read bytes: " + str(discarded_bytes))
TimeoutError: Can't get to raw prompt. Read bytes: b'rl\x00l'
```

The dev board works fine with a standard serial terminal (Putty on Windows 7) and PyCharm's Micropython REPL.   I tried it with both an ESP-12E module and a ESP-07S module and got the same result so I'm pretty sure it has something to do with the dev board.

FWIW, the Thonny REPL works fine with other ESP8266 Nodemcu CP2102 based modules, and the original PyBoard (with thonny-pyboard plug-in).  I'm using version 6.7.6 of the Silabs CP210x VCP driver, and the MicroPython (generic) interpreter.

aivarannamaa, 2019-10-16:
Thank you for the report! I ordered this board and an ESP-07S. Hopefully I can reproduce the problem once they arrive.

aivarannamaa, 2019-10-29:
Thonny choked on the garbage that was output just after connecting (probably it was meant to be read with different baud rate). I now made the reader more robust:

```
![image](https://user-images.githubusercontent.com/1802/67813199-9c49a580-faa9-11e9-8108-9309a987b321.png)

I wonder how Putty is able to discard this garbage... It does show a piece of junk, but much smaller:

![image](https://user-images.githubusercontent.com/1802/67813258-c7cc9000-faa9-11e9-8d90-d46ee7554a27.png)

closed 2019-10-29 by aivarannamaa

JennaSys, 2019-10-31:
Just fyi, I tested the update and it now works fine in my environment as well.  Thank you for the quick fix!

commit f07fb4e1 (2019-10-29): FIXED #946: Handle broken UTF-8 codepoints in MP output