thonny/thonny #950: Warn if "thonny" package comes from user site packages

aivarannamaa, 2019-10-19:
TODO: how to make this work such that the warning is given also when older Thonny versions are loaded?

aivarannamaa, 2019-10-19:
Make bundled Thonny run with [PYTHONNOUSERSITE](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONNOUSERSITE) and add user site to path only for loading plugins?

Ivoz, 2019-10-27:
What's the problem with it being installed there?

aivarannamaa, 2019-10-27:
The motivation came from this: https://groups.google.com/forum/#!msg/thonny/FOt7M-qy-Fg/zTQAH2AsBwAJ

I'm not entirely sure, whether such warning would be the best solution.

Ivoz, 2019-10-27:
Perhaps also including the location Thonny is currently running from in the About screen could also help? 🤔 

That kinda seems like a totally valid location, to put python packages, if indeed someone is installing Thonny as a python package. For a simple "user" program, putting it in user-site doesn't require any administration privileges, etc.

Although both this problem (hopefully?) and the above case I just described could be quite niche... another approach could be if it wasn't too much effort to put a script in the installer looking for other "normal" locations that a Thonny could be, and warning the user? Non-trivial problem.

aivarannamaa, 2019-10-30:
Thank you for the input! I'll postpone this for now, considering possible changes in Thonny's `sys.path` in version 4.0 (#718)

aivarannamaa, 2022-02-09:
Not relevant after #718

closed 2022-02-09 by aivarannamaa