thonny/thonny #953: Consider add configurable menu item

sengpw, 2019-10-22:
I'd like to use "thonny" to write micropython code for ESP32. Seems to be very comfortable, but see no way to check the code with "flake8" or something like that before code is downloaded to ESP32. Think it would be great to have one (or more) configurable menue/toolbar items to call external tools. Attached see how this can be configured in "Geany-Editor/IDE".
[image]

aivarannamaa, 2019-10-23:
Thank you for the suggestion! I'll consider this for future releases.

Did you know that if you have flake8 installed (eg. from Tools => manage packages), you can run it from thonny shell with `>>> !flake8 my_script.py` ?

sengpw, 2019-10-23:
thanks for response and hint.
In shell "flake8" does not work:

>>>
>>> !flake8 main1.py
Unknown command: execute_system_command
>>>

When I do:

Tools --> Open system shell -->

user@vPad ~/temp $ flake8 main1.py
main1.py:28:80: E501 line too long (98 > 79 characters)
user@vPad ~/temp $

It works as expected.

What is wrong with "execute_system_command"?
Tools --> Manage plug-ins --> shows menu attached.
[image]

aivarannamaa, 2019-10-24:
Oops, I forgot that this is not available with MicroPython back-end. I'll add it there as well.

aivarannamaa, 2019-10-27:
I made system commands work in MicroPython backend. Note that current directory for the commands is the same directory as selected in the local files part of Files view.

I'll postpone the implementation of configurable menu items.

sengpw, 2019-10-28:
checked out mater branch (2019-10-28), installed (sudo python3 setup.py install) and tried to run thonny.
Errors on program start and on stop/restart backend see attached screenshot.

[image]

aivarannamaa, 2019-10-29:
Oops, I forgot about Python 3.5. I'll fix it later.

sengpw, 2019-11-26:
On version 3.2.3 shell command works. Very nice - thank you!

aivarannamaa, 2019-11-26:
Let's keep it open, because configurable menu item is still a good idea.

sengpw, 2019-11-27:
After using Thonny/MicroPython now for some day's (I'm new to Python) I really appreciate using !flake8... It makes much sense checking syntax/style in before. Menu icon "Run Tool" or "flake8 current script" would be very nice, even better if click on error/warning would make a jump to the correct line in the editor.

kr-g, 2021-02-26:
the _`!cmd`_ feature is nice. 
but calling `!black somepath` will crash thonny. 
`!flake8` works, also `!ls` or such a like