thonny/thonny #954: Open new shell tab?

boardtc, 2019-10-22:
The Shell tab has context sensitive menu, Close, Close others & Close all, which suggests there is a way to open a second and subsequent shells. I can't find an option to do this, what am I missing?

aivarannamaa, 2019-10-22:
This is the result of a general solution of managing tabs in Thonny. It makes sense if you open "Program tree" from view menu.

At every moment Thonny maintains single background process and single shell.

closed 2019-10-22 by aivarannamaa

boardtc, 2019-10-23:
Ok. I have not been able to see the contents of Program Tree yet. I have reported a bug on that. I assume you mean that program tree is other tab allowed open then since you say only one shell is possible.

aivarannamaa, 2019-10-24:
> I assume you mean that program tree is other tab allowed open then since you say only one shell is possible.

That's right