thonny/thonny #955: Window panes render with lines jamed up on each other

boardtc, 2019-10-23:
Is there anything that can be done about the lines in panes jamming up against each other? I can manage in variables but Program Tree is a disaster.
[image]
[image]

aivarannamaa, 2019-10-23:
Please try different scaling rate in Tools => Options => General

boardtc, 2019-10-23:
Neither of the other 2 options made a difference, extra or automatic, I restarted Thonny each time.

aivarannamaa, 2019-10-24:
I'll see what I can do. Duplicate of #933

closed 2019-10-24 by aivarannamaa

boardtc, 2019-10-26:
Same issue on Variables pane. I added to [#933 ](https://github.com/thonny/thonny/issues/933#issuecomment-546543468)