thonny/thonny #957: the eccoding of thonny's shell

rnLIKEm, 2019-10-23:
please look at the picture below immediately.
[image]

rnLIKEm, 2019-10-23:
it's seems that the encoding is not utf-8.
and when I start thonny,the cursor of  shell shine in front of >>> not (>>> cursor(shine))(this is a little problem)

aivarannamaa, 2019-10-23:
Thank you for the report!
Please post the text for a (preferably as small as possible) program, which replicates this problem.

rnLIKEm, 2019-10-23:
'''
# coding:utf-8
from multiprocessing import Pool
import os

def foo(name):
    print('你好!')

if __name__ == '__main__':
    p = Pool(2)
    for i in range(4):
        p.apply_async(foo, args=(i,))
    p.close()
    p.join()
    print('End')
'''

rnLIKEm, 2019-10-23:
` ` ` # coding:utf-8
from multiprocessing import Pool
import os

def foo(name):
    print('你好!')

if __name__ == '__main__':
    p = Pool(2)
    for i in range(4):
        p.apply_async(foo, args=(i,))
    p.close()
    p.join()
    print('End')
` ` `

rnLIKEm, 2019-10-23:
```# coding:utf-8
from multiprocessing import Pool
import os

def foo(name):
    print('你好!')

if __name__ == '__main__':
    p = Pool(2)
    for i in range(4):
        p.apply_async(foo, args=(i,))
    p.close()
    p.join()
    print('End')
```

aivarannamaa, 2019-10-23:
I can't reproduce this in Linux. I'll try in my Windows computer later.

rnLIKEm, 2019-10-23:
good luck

rnLIKEm, 2019-10-23:
[image]
And,without ** p.join() ** statement .run this program,wait for monent ,the main process and the child process is over,run it again will get an exception.

closed 2019-10-26 by aivarannamaa

rnLIKEm, 2019-10-26:
Hey,did you fix the second problem? shall I open it as another issue?

aivarannamaa, 2019-10-26:
Sorry, I forgot the second one.

I now looked into it and I don't know how to fix it. Each time a script is run, Thonny starts a new python subprocess. It looks like something is not cleaned up fast enough -- when I add a `sleep(0.1)` before starting new subprocess, everything runs fine (`sleep(0.01)` is not enough). I don't want to solve it with this hack, because it would slow down the programs which are not using multiprocessing, and it's not guaranteed that 0.1 seconds is always enough.

Is there a reason you don't want to use `join`?

rnLIKEm, 2019-10-26:
No,I don't have.
I am a python novice,I like use thonny to learn python, because it looks very simple,and everything looks so clearly.
so I try every python syntax in thonny,then I encountered this problem,then I try the same code in IDLE and spyder(to avoid I used some wrong python syntax) then I realized this a bug,so I report it.

rnLIKEm, 2019-10-26:
these is some problem with spyder,even it use ipython kernal .it can't get subprocess what have printed.
and can't know what encoding it use in subprocess.just like thonny before

rnLIKEm, 2019-10-26:
If you use `sleep(0.1)`,it will slow the shell speed,I have test the speed of several terminals(IDLE, thonny shell, MS cmd, git bash,)and I get thonny shell just a little faster than IDLE, and git bash is very very fast.

rnLIKEm, 2019-11-01:
About the second bug,maybe there is something wrong with multiprocessing module.the developer of 
ipython said.see [https://github.com/ipython/ipython/issues/11929#issuecomment-548601116](url)

aivarannamaa, 2019-11-01:
That's good news for me :)

commit be5e141d (2019-10-26): FIXED #957: Wrong encoding in multiprocessing output