thonny/thonny #961: Thonny 3.2.1 under Catalina 10.15.1

tatecapone, 2019-10-24:
First, the application does not start without the `sudo xattr -rd com.apple.quarantine` command in the terminal. Secondly, once thonny is open, it is impossible to close it ... and I have to force the exit through the menu bar-> apple-> force the exit.
Thanks!!

aivarannamaa, 2019-10-24:
Thank you for the report! This seems to be somewhat related to but different from #813.

Did you install Thonny in Catalina or it was installed under previous OS version?

tatecapone, 2019-10-24:
Thank you for the response!
I just installed thonny, because I found the command `sudo xattr -rd com.apple.quarantine` since before I couldn't ... so I already had Macos Catalina 10.15.1 running ...

tatecapone, 2019-10-24:
the shell works perfect..there seems to work very well ... for now ...

tatecapone, 2019-10-24:
oooops!  Thonny close perfectly with the x button....
I think you can close this issue.
really, really sorry for this wasted time.
Thanks!!

aivarannamaa, 2019-10-24:
I'll leave it open for now, because it contains useful information.

tatecapone, 2019-10-24:
So, until now, the only problem is that it does not open without the above-mentioned command via the terminal.
Ok...Thanks aivarannamaa!!
Best regards!

aivarannamaa, 2020-02-05:
Please try, whether https://github.com/thonny/thonny/releases/download/v3.2.6/thonny-3.2.6.pkg fixes the problem. 

#79

aivarannamaa, 2020-02-06:
Ouch, it doesn't work in a clean system (ie. without preinstalled Python). I'll deal with it.

aivarannamaa, 2021-01-14:
Assuming this got fixed after introducing pkg installer

closed 2021-01-14 by aivarannamaa