thonny/thonny #963: run a multiprocess program without `self.join()` twice, will get an exception.

rnLIKEm, 2019-10-26:
see the picture below
[image]
the code below:
```
# coding:utf-8
from multiprocessing import Process

def foo():
    pass

if __name__ == '__main__':
    p = Process(target=foo, )
    p.start()
    print('End')
```

closed 2019-10-26 by rnLIKEm

aivarannamaa, 2019-10-30:
Reminder for myself: At the moment I don't know how to fix it. Each time a script is run, Thonny starts a new python subprocess. It looks like something is not cleaned up fast enough -- when I add a sleep(0.1) before starting new subprocess, everything runs fine (sleep(0.01) is not enough). I don't want to solve it with this hack, because it would slow down the programs which are not using multiprocessing, and it's not guaranteed that 0.1 seconds is always enough.

rnLIKEm, 2019-11-10:
did you noticed when a subprocess run over,it will not dead?
it seems that after call it ,it will dead.I have made some pictures,see:
[image]
[image]
[image]
[image]

aivarannamaa, 2019-11-10:
This matches with my tests -- it looks like a process remains alive after main thread finishes.

rnLIKEm, 2019-11-13:
it's seems that which is waitting for something.
maybe it's waitting for me to enter `p` in the shell then press `enter`