thonny/thonny #964: Internal error when attempting to clear shell

willbprog127, 2019-10-27:
Greetings!

After using Thonny 3.2.1 for a few hours, I tried to clear the Shell display by right-clicking in the Shell then choosing 'Clear'.  The following internal error popped up:

```
Traceback (most recent call last):
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/usr/local/lib/python3.7/dist-packages/thonny/shell.py", line 1235, in _clear_shell
    self._clear_content(end_index)
  File "/usr/local/lib/python3.7/dist-packages/thonny/shell.py", line 1308, in _clear_content
    btn_pos = float(self.index(btn))
ValueError: could not convert string to float: 
```
This is running the Thonny from `pip` installation.
* Debian 10 amd64
* Python 3.7.3

Thank you! :smiley:

aivarannamaa, 2019-10-27:
Thank you for the report!

I'm not able to reproduce this. Can you do it again or was it a single case?

willbprog127, 2019-10-27:
> Can you do it again or was it a single case?

After 'something happened' (I don't know what specifically) it would do the error above continuously.  I could not clear the shell after that point unless I closed Thonny and tried again.

aivarannamaa, 2020-03-16:
Duplicate of #1091

closed 2020-03-16 by aivarannamaa