thonny/thonny #965: Assistant sometimes steals focus from code editor right after running program

willbprog127, 2019-10-27:
Right after running a Python program within Thonny, when you start to edit something in the code editor, the Assistant sometimes steals the focus when it refreshes.  You have to then click back into the code editor to continue.

This slows down editing and debugging of whatever program you're working on and is very distracting.  This happens five out of ten times for me.

This is running the Thonny from `pip` installation.

* Debian 10 amd64
* Python 3.7.3

Thank you! :smiley:

Ivoz, 2019-10-27:
Do you have an example program you could paste that demonstrates this in your case?

aivarannamaa, 2019-10-27:
Thank you for the report!

I can't reproduce it in Debian 10-based Raspbian. Are you using Thonny+Python binary bundle or the one from your distro reposiory? Which version?

A sample program would be useful.

aivarannamaa, 2019-10-27:
> I can't reproduce it in Debian 10-based Raspbian. Are you using Thonny+Python binary bundle or the one from your distro reposiory? Which version?

Sorry, I now noticed "This is running the Thonny from pip installation."

willbprog127, 2019-10-27:
Thank you for the comments.

I do not have a program I can share at the moment, but I'll see what I can come up with.

Thanks! :+1:

Ivoz, 2019-11-03:
You think it may be a lag issue? you have a large code file, the assistant takes a while to update, and when it does it steals focus?

willbprog127, 2019-11-03:
> You think it may be a lag issue? you have a large code file, the assistant takes a while to update, and when it does it steals focus?

Yes, it was a very large amount of code.  It appears (but not quite certain) that the focus is stolen after the assistant updates.

aivarannamaa, 2021-02-26:
Closing for now, as I don't how to proceed. Please reopen if you find a way to reproduce.

closed 2021-02-26 by aivarannamaa