thonny/thonny #966: Thonny encounters "internal error" in programs calling exit()

Phoemuex, 2019-10-27:
I am running Thonny 3.2.1 on Windows 8.1 (64 bit), using Python 3.7.4 (32 bit) and Tk 8.6.9 (downloaded and installed the bundle from the website).

I have a curious issue regarding programs that call exit(). These work fine usually, but in connection with importing tkinter, they seem to cause issues.

Consider the following two-line program

```
import tkinter
exit()
```

Saving this program as "test.py" and running it (via F5) causes Thonny to display the following "internal error":

```
Traceback (most recent call last):
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 749, in callit
    func(*args)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 432, in _poll_vm_messages
    if self._pull_vm_messages() is False:
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 440, in _pull_vm_messages
    msg = self._proxy.fetch_next_message()
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1141, in fetch_next_message
    msg = super().fetch_next_message()
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 942, in fetch_next_message
    self._store_state_info(msg)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 988, in _store_state_info
    self._update_gui_updating(msg)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1015, in _update_gui_updating
    self._loop_gui_update(True)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1021, in _loop_gui_update
    self.send_command(InlineCommand("process_gui_events"))
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 791, in send_command
    self._send_msg(cmd)
  File "C:\Users\Felix\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 795, in _send_msg
    self._proc.stdin.flush()
OSError: [Errno 22] Invalid argument
```

NOTE: The same does not happen if one omits the `import tkinter`!

NOTE 2: This error is not completely reproducible, sometimes the program finishes without the internal error, but at least after running the program three times in a row, I have until now ALWAYS been able to reproduce the problem, in ~15 tries.

aivarannamaa, 2019-10-30:
Thank you for the report!

I was able to reproduce it only once (in about 30 trials), but I think I found the solution. Please open the issue again if it persists in upcoming Thonny 3.2.2

LordzShadow, 2022-09-01:
This issue seems to persist in Thonny 4.0.0

running exit() the program does not end and clicking on the shell window and pressing enter, Thonny crashes.
```
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\tktextext.py", line 243, in dispatch
    return fun(event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 1012, in perform_return
    EnhancedTextWithLogging.perform_return(self, event)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\tktextext.py", line 468, in perform_return
    self.insert("insert", "\n")
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 974, in intercept_insert
    self._try_submit_input()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 1163, in _try_submit_input
    self._submit_input(submittable_text)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\shell.py", line 1320, in _submit_input
    get_runner().send_program_input(text_to_be_submitted)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 309, in send_program_input
    self._proxy.send_program_input(data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1057, in send_program_input
    self._send_msg(InputSubmission(data))
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\running.py", line 1050, in _send_msg
    self._proc.stdin.write(serialize_message(msg) + "\n")
OSError: [Errno 22] Invalid argument
```

[image]

aivarannamaa, 2022-09-01:
Thanks for reporting this! I'll try to fix it for Thonny 4.0.1

aivarannamaa, 2022-09-11:
Fixed under #2433.

closed 2022-09-11 by aivarannamaa

vandega, 2022-11-08:
this error goes in 4.0.1. after easy code. 

`Traceback (most recent call last):
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 839, in callit
    func(*args)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\ui_utils.py", line 2244, in poll
    on_completion(proc, out_lines, err_lines)
  File "C:\Users\User\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\mypy\__init__.py", line 110, in _parse_and_output_warnings
    logging.error("Can't parse MyPy line: " + line.strip())
NameError: name 'logging' is not defined

`

aivarannamaa, 2022-11-10:
@vandega, this was fixed in #2491 and the fix will be available in the next version. In the meantime you can get rid of this error by going to "Tools => Options => Assistant" and unchecking "Perform MyPy checks".

commit f8c02a68 (2019-10-30): FIXED #966: Thonny encounters "internal error" in programs calling exit()