thonny/thonny #968: Flatpak build

LeandroStanger, 2019-10-29:
Please, consider supporting [Flatpak](http://docs.flatpak.org/en/latest/introduction.html#reasons-to-use-flatpak) and publishing thonny on [Flathub](https://github.com/flathub/flathub/wiki/App-Submission).

aivarannamaa, 2019-11-02:
I prefer to focus on core development myself, but I welcome anyone else who wants to do this.

jwillikers, 2021-07-13:
I'll look into this, though I know a Thonny maintainer should make the actual PR to Flathub since that will grant the submitter write-access to the Flathub Thonny repository.

aivarannamaa, 2021-07-13:
@jwillikers, if you prepare the code, then I can submit the PR if required.

jwillikers, 2021-07-17:
I've started working on a `Devel` Flatpak manifest for building Flatpaks locally.
The Flathub repository will end up hosting a pretty similar manifest.
There's some issues that need worked out to build the Python dependencies, so I'll start poking upstream / checking back to see if they get resolved when I have time.

PR #1900 (2021-07-22): Flatpak
This MR is to add a development Flatpak manifest to to the Thonny repository.
This work will allow developers to build Flatpaks locally with the latest source and makes it easy to integrate into CI workflows if desired.
The work here should be easy to transfer to a repository on Flathub.

At the moment, the Flatpak Pip Generator is having difficulties with a couple of dependencies, so this isn't working yet.
The current issue appears to stem from Pylint 2.8.3 and occurs when flatpak-builder is building the Python dependencies.

Fixes #968
closed 2021-07-22 by aivarannamaa