thonny/thonny #974: Thonny Backend Terminated and wont let me Run/Restart (help please :c)

Gianowo, 2019-11-07:
This is my file pathway and the error message I recieve upon opening the program.
I'm clueless about what led to this but this happened spontaneously.
I'd appreciate if someone could find a fix because i've searched far and wide and no dice.
Also, Thonny is really user friendly and it'd be a shame if I had to migrate to another IDE.

Traceback (most recent call last):
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1705, in __call__
    return self.func(*args)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 406, in cmd_stop_restart
    self.restart_backend(True)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 524, in restart_backend
    self._proxy = backend_class(clean)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1160, in __init__
    super().__init__(clean, executable)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 973, in __init__
    super().__init__(clean, executable)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 718, in __init__
    self._start_background_process(clean=clean)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 728, in _start_background_process
    env = get_environment_for_python_subprocess(self._executable)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1300, in get_environment_for_python_subprocess
    overrides = get_environment_overrides_for_python_subprocess(target_executable)
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\site-packages\thonny\running.py", line 1347, in get_environment_overrides_for_python_subprocess
    target_executable, this_executable
  File "C:\Users\Gianc\AppData\Local\Programs\Thonny\lib\genericpath.py", line 96, in samefile
    s1 = os.stat(f1)
FileNotFoundError: [WinError 3] The system cannot find the path specified: ''

aivarannamaa, 2019-11-07:
Please post your configuration.ini (Open Thonny => Tools => Open Thonny Data folder)

Gianowo, 2019-11-07:
@aivarannamaa It turns out i didnt need to do that,
I waited a day and everything went back to normal again, I don't know why but Thonny miraculously reinstalled itself ;after my attempts of opening Thonny an error message popped up which also brought up the install window.
All the user settings are reset but its working good as new.

aivarannamaa, 2019-11-10:
I'm glad you got it working!

closed 2019-11-10 by aivarannamaa

Gianowo, 2019-11-11:
My apologies,
All it took was a quick look into the program's path in file explorer and sending the Thonny appdata and program files into the recycle bin, once that was finished ,all it needed was a quick reinstall and voila~ everything was in working order again :)