thonny/thonny #976: boot.py definitions lost when using Stop/Restart backend button

adzierzanowski, 2019-11-07:
Consider a simple micropython `boot.py` script:

```python
import time
hello = 'hello, world'

while True:
    print(hello)
    time.sleep(1)
```

When interrupting within the shell with ctrl+C, it just throws an exception and I still can use the `hello` variable in the interpreter. But when I use Stop button instead, the backend restarts and the access to `hello` is lost because it is no longer defined.

aivarannamaa, 2019-11-07:
Thank you for the report!

I wanted to emulate the behavior of CPython backend, where Stop/Restart really cleans the environment (by restarting the process and not running anything).

In MicroPython back-end I clear the environment manually, but it looks like this approach doesn't suit here.

aivarannamaa, 2020-10-28:
Fixed in 3.3

closed 2020-10-28 by aivarannamaa