thonny/thonny #977: Debug(Nicer) aborts with AssertionError

andreas0806, 2019-11-07:
The following small code snippet is causing  thonny 3.2.1 to stop debugging when running the "nicer" debugger, the "faster" works without problem

 ```
import csv

sCSV = '"ID";"Camera_ID_L";"x_L";"y_L";"Camera_ID_R";"x_R";"y_R"' 
pos = csv.DictReader(sCSV, delimiter=';', quotechar='"', quoting=csv.QUOTE_NONNUMERIC)
gps_data_2D = {}
for row in pos :
    print( "1" )         
```
After exiting the for loop the debuggin process is aborted with :

`Traceback (most recent call last):
  File "C:\Program Files\JOANNEUM RESEARCH\ImpactTools\lib\site-packages\thonny\backend.py", line 1575, in _trace
    return self._trace_and_catch(frame, event, arg)
  File "C:\Program Files\JOANNEUM RESEARCH\ImpactTools\lib\site-packages\thonny\backend.py", line 1637, in _trace_and_catch
    assert last_custom_frame.event.startswith("before_")
AssertionError`

Best regards,
Andreas

aivarannamaa, 2019-11-10:
Thank you for the report! I'll investigate it later

aivarannamaa, 2019-12-06:
Fix will be in next release.

closed 2019-12-06 by aivarannamaa