thonny/thonny #979: Such an error regularly flies out when you press the stop button!

AlexanderSemenov1, 2019-11-09:
Crash in backend
Traceback (most recent call last):
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 100, in __init__
    self._prepare(clean)
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 110, in _prepare
    self._interrupt_to_raw_prompt()
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 257, in _interrupt_to_raw_prompt
    raise TimeoutError("Can't get to raw prompt. Read bytes: " + str(discarded_bytes))
TimeoutError: Can't get to raw prompt. Read bytes: b''
Traceback (most recent call last):
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 100, in __init__
    self._prepare(clean)
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 110, in _prepare
    self._interrupt_to_raw_prompt()
  File "/home/alex/apps/thonny/lib/python3.7/site-packages/thonny/plugins/micropython/backend.py", line 257, in _interrupt_to_raw_prompt
    raise TimeoutError("Can't get to raw prompt. Read bytes: " + str(discarded_bytes))
TimeoutError: Can't get to raw prompt. Read bytes: b''

Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

aivarannamaa, 2019-11-10:
Which device are you using? Which Thonny version?

AlexanderSemenov1, 2019-11-10:
This happens on one of my ESP32. I checked other boards and there is no such mistake on them. I also tried ESP8266 and they also didn’t have such an error. I am using Tonny 3.2.1, Linux Mint 19.2 (64-bit), Python 3.7.4 (64-bit), Tk 8.6.9. I assume this problem is related to my ESP32 board as other boards do not cause such an error.
In the terminal, using the dmesg command, data from usb is read correctly as for my other devices on cp2102.
Thank you for responding so quickly! :)

aivarannamaa, 2019-11-10:
Please check whether version 3.2.3 works better with this device: https://github.com/thonny/thonny/releases/tag/v3.2.3

AlexanderSemenov1, 2019-11-25:
Hi!. I installed version 3.2.3, the problem remains. I found that this problem is only on the board with cp2102.
Boards with cp2104 and ch340 chip work fine!

aivarannamaa, 2019-11-25:
Can you get to the REPL using some other tool (eg. `screen` or `rshell`)?

Can you give me a link where one could buy this problematic board?

AlexanderSemenov1, 2019-11-28:
Hey.
I used to use the Windows 10 Home system and the EsPy program (https://github.com/jungervin/EsPy), there were no such problems. But on OC Linux, this program does not work. And I chose your program.
I tried connecting to esp32 in PuTTY and the connection is working fine.
In Thonny, the first connection is excellent. But after pressing the stop / reboot button, errors appear. And only from the third or fourth time by pressing the stop / reboot button does it turn out to be connected to the board again. And this only happens on a board with a cp2102 chip.

I bought a board here https://en.aliexpress.com/item/32844181786.html?spm=a2g0s.9042311.0.0.274233edxSBz7A

aivarannamaa, 2019-11-28:
> I bought a board here https://en.aliexpress.com/item/32844181786.html?spm=a2g0s.9042311.0.0.274233edxSBz7A

This link gives me 404

AlexanderSemenov1, 2019-11-28:
https://ru.aliexpress.com/item/32844181786.html

AlexanderSemenov1, 2019-11-28:
ok?

aivarannamaa, 2019-11-28:
Now I got it. The first link would work if you remove `en.`

I'll order this and hopefully I can reproduce the error once it arrives (next year)

AlexanderSemenov1, 2019-11-28:
I’ve tried a new reset method by combining the buttons Ctrl + D and stopping the program Ctrl + C, everything works fine! Since the established connection does not break.
Maybe there is an opportunity to create buttons on the taskbar with these functions?
And to assign the stop / reboot button as connect / disconnect?

aivarannamaa, 2019-11-28:
Please try following: 

* In Thonny select "Tools => Open Thonny program folder"
* load `plugins/micropython/backend.py`
* in method `clean` after `if clean:` insert following line `self._soft_reboot(True)`
* Save and try Stop/Reset button again

Does it work better like this?

aivarannamaa, 2021-02-26:
Please try again with latest Thonny versions and reopen if the problem remains.

closed 2021-02-26 by aivarannamaa