thonny/thonny #980: Thonny debugger won't respond

MadCoder45, 2019-11-09:
I am accessing the Thonny IDE through my raspberry pi 4. I am trying to run a simple buzzer code, my program is written below. When I go to click the debug button next to run at the top of my screen, the button goes dim for a second and returns to normal without proceeding to debug my code. Does anyone know why this is happening?

My circuit looks the same as it does on this website: [(https://projects.raspberrypi.org/en/projects/physical-computing-with-scratch/11)]
(Only use the link for reference of how my circuit looks, not the code please.)

Any help is great, thank you.

`from gpiozero import Buzzer
from time import sleep

buzzer = Buzzer(17)

buzzer.on()
sleep(1)
buzzer.off()
while True:
buzzer.on()
sleep(1)
buzzer.off()
sleep(1)
buzzer.beep()`

aivarannamaa, 2019-11-10:
Thank you for the report! Does the program behave properly with "Run" command?

MadCoder45, 2019-11-10:
> Thank you for the report! Does the program behave properly with the "Run" command?

No, the 'run' function was unresponsive. I figured out the problem... Turns out that I needed a while True loop, and to indent the lines of code. Thank you though. My final code is written below.

`from gpiozero import Buzzer
from time import sleep

buzzer = Buzzer(17)

     while True:
     buzzer.on()
     sleep(1)
     buzzer.off()
     while True:
     buzzer.on()
     sleep(1)
     buzzer.off()
     sleep(1)
     buzzer.beep()`

closed 2019-11-10 by MadCoder45