thonny/thonny #985: Thonny 3.2.3 Windows installer executable untrusted

Ivoz, 2019-11-12:
Did something go wrong with signing?

[image]

[image]

aivarannamaa, 2019-11-12:
It is now signed with a new identity and it takes time (and many installations) before Windows Defender starts to trust it (https://stackoverflow.com/questions/48946680/how-to-avoid-the-windows-defender-smartscreen-prevented-an-unrecognized-app-fro).

I was hoping that it is already trusted ...

It's a real shame that a signed app can cause more scary dialog than an unsigned one.

I'll put a notice about this to the download box.

Ivoz, 2019-11-13:
Dayum; I don't know the details of MS' trust process for executables, but I thought maybe some process had gone wrong for you with it's release, that was fixable.

Is this trust building needed for every new executable you produce? :( Hope not. Otherwise I wonder if maybe an EV cert could be something sponsored (maybe by PSF or etc?) in the future that you could use.

Good luck with it!

closed 2019-11-13 by Ivoz

aivarannamaa, 2019-11-13:
I believe the trust will be gathered for the signer, not for the executable. 

Regarding the EV cert -- I considered this, but I was afraid of the hassle that the validation would bring along.

Ivoz, 2020-02-13:
Good news, 3.2.6 recently I saw no untrusted label.

aivarannamaa, 2020-02-13:
Thanks for letting me know, that's great news!