thonny/thonny #988: Thonny Greek Translation issues

niplessa, 2019-11-14:
Hello. 
I'm translating Thonny in Greek language as a project for my university.

I have these issues:

1) "**Options**" window is too narrow and doesn't show all the tabs . 
Screenshot:
[image]

2) There is a "**Device**" tab that doesn't exist in POeditor and i can't translate it. 
My supervising professor suggested that it is deprecated and won't be used in future versions so it is not on POeditor on purpose.
Screenshot:
[image]

3) My professor is using Thonny on MAC. In his version of Thonny there is a menu "**Thonny**" left of the "File" Menu. I can't see the "Thonny" menu neither in 3.2.1 or 3.2.3 editions in Windows.
Screenshot:
[image]

Apart from the above mentioned issues, our translation is ready. Although i don't know how to make a pull request with my .po & .mo files.

Thank you very much,
Nikos Plessas

niplessa, 2019-11-16:
anyone?

aivarannamaa, 2019-11-16:
> "Options" window is too narrow and doesn't show all the tabs .

Thanks for reporting, I'll make it wider.

> There is a "Device" tab that doesn't exist in POeditor and i can't translate it.
> My supervising professor suggested that it is deprecated and won't be used in future versions so it is not on POeditor on purpose.

That's correct.

> I can't see the "Thonny" menu neither in 3.2.1 or 3.2.3 editions in Windows.

"Thonny" menu is supposed to exist only on macOS, as all apps on this OS traditionally have this kind of "Application menu".

> Apart from the above mentioned issues, our translation is ready.  Although i don't know how to make a pull request with my .po & .mo files.

I already took the translation from Poeditor and included it in 3.2.3: https://github.com/thonny/thonny/releases/tag/v3.2.3
You don't need to do anything else.

closed 2019-12-06 by aivarannamaa

aivarannamaa, 2019-12-07:
It turned out the dialog becomes too wide when I try to fit all Greek labels. I'll solve this issue later when I refactor the dialog (#730)