thonny/thonny #990: Always store a local copy when save a file on a MicroPython device.

jedie, 2019-11-15:
I think problem https://github.com/thonny/thonny/issues/531 isn't well solved yet.

Thonny should always store a local copy when save a file on a MicroPython device. It's difficult to manage the sources in the git.

I had the case that I had programmed something wrong and no communication with the device was possible anymore. So I couldn't download my files anymore.
(I recover my source files by using esptool.py read_flash, see: https://forum.micropython.org/viewtopic.php?f=2&t=7246 )

aivarannamaa, 2019-11-15:
I'll think about this. 

(I assume you know that Thonny still supports running scripts from local computer and uploading them when they are ready.)

jedie, 2019-11-15:
> I assume you know that Thonny still supports running scripts from local computer and uploading them when they are ready.

Yes, but it's a slower workflow, isn't it?

Really cool, if `F5` will save the file locally and run the script on device, too.

aivarannamaa, 2019-11-15:
> Really cool, if F5 will save the file locally and run the script on device, too.

That's what it currently does for local files.

A related topic -- Thonny now also supports uploading whole directories. It is less efficient than sync, though, as it doesn't check whether the target file is identical (eg. by hashing). I suppose if I implemented this check, it would be a step towards what you need.

closed 2021-02-26 by aivarannamaa

mitteKhanh, 2021-02-28:
I was also expecting this behaviour - an option to keep local and on-device files in sync would be great!

aivarannamaa, 2021-02-28:
@mitteKhanh, I have plans for this, but it will syncing from the local disk to the device, not vice versa.

mitteKhanh, 2021-03-01:
sure that works too. good to hear that :)

briceparent, 2021-03-02:
@aivarannamaa That would be great
Until now, I haven't really found an easy way to have a local copy of the files I'm working on. Whenever I work on a single file, execution is enough, but when I develop libraries, in other files, and test them by importing them and using the functionality from main script (which I just execute), I have to save them locally manually once it works. And if there is still any issue, I update it and save it to the board, etc. And I should keep track of the changes I've made to every file before I can commit the files to git.

cobra18t, 2022-01-26:
This is devastating. I had been working directly on the device since I was interacting with hardware as part of development. I had a hardware failure on my board and now all that work is lost. It would have been amazing if there was a local copy that I could recover.

rapzak, 2023-02-13:
++ For save local copy... closed the editor, and the file in the device was 0bytes from last save :(....

Nice to have a copy...

aivarannamaa, 2023-02-13:
I still recommend keeping your master copies on local disk and copy them to the MicroPython filesystem after testing. Please note that you can test your script on target device without saving it there.

rapzak, 2023-02-13:
Thank, i figured that out between i wrote and you wrote. On My wb55 i have often issues to write the file to the device... I May guess it is in the flash writing in the port...

DdOtzen, 2023-10-03:
> I still recommend keeping your master copies on local disk and copy them to the MicroPython filesystem after testing. Please note that you can test your script on target device without saving it there.

That just don't work when working on libraries, or just projects that is more than one file.