thonny/thonny #993: No file dialogue when opening or saving files.

laconic75, 2019-11-17:
I'm running Thonny 3.2.3 on an Arch Linux box and am getting no file dialogue when opening or saving a file. I've tried installing from a release tar.gz and from 'pip install' and still have the same problem. 

I would like to give you more information, but I see no errors or really anything in the log files. I've tried setting "debug_mode = True" in the configuration.ini to try and get more information, but I still see nothing when I try to open or save a file. 

Any help you could give me with trying to track this down would be appreciated.

Thanks,

aivarannamaa, 2019-11-17:
Which Thonny version and which Arch version are you using?

Do you have `zenity` installed? Which version? (enter `zenity --version` in Terminal)

laconic75, 2019-11-17:
Thanks for the reply. Mentioning `zenity` pushed me in the right direction. 
It was just a peculiarity of my environment. I'm running KDE and am using a `zenity` replacment called [qarma](https://github.com/luebking/qarma).  It hadn't been updated in a while. Getting the latest version with [AUR and git](https://aur.archlinux.org/packages/qarma-git) was able get it working.

closed 2019-11-17 by laconic75