thonny/thonny #995: Feature request: drag to open file

LowToneMoFo, 2019-11-19:
I would like to request/suggest a quality of life feature.

Currently it is not possible to drag a file into Thonny to open it. It has to be opened from the menu. I use Thonny in an educational environment and when reviewing my students projects I have to open a lot of different files in a short time. It would be ideal to drag in all files that I want to review to have Thonny open them in different tabs.

aivarannamaa, 2019-11-19:
It's quite troublesome thing to implement, as Tkinter by default doesn't support drag-n-drop.

It seems to be possible to enhance the Tk in Python+Thonny bundles, though (https://stackoverflow.com/questions/25427347/how-to-install-and-use-tkdnd-with-python-tkinter-on-osx).

Have you tried opening files from Thonny's file browser (View => Files)?

LowToneMoFo, 2019-11-20:
The View -> Files on is a great suggestion. I never noticed this option. It basically fills in the same need.

Cheers!