thonny/thonny #998: Only show local variables in variable view

cspaier, 2019-11-22:
Hello,

At the moment, the variable view is showing all variables.

I would like to add a setting so that it only show variables from the executed script. This would hide all variables that came from imports.

The motivation is related to teaching. If a teacher ever does a wild import `from lib import *`, the variable view become hard to use at the moment.

I noticed #174 and it could already help but I think a setting as described would suit our teaching best.

Would you be ok with such a PR?

aivarannamaa, 2019-11-22:
If current behavior remains default, then I don't mind including it. How do you plan to implement it? 

Would it be easy to target #174 as well and introduce a three-valued configuration variable for imported names: ["No special treatment", "Show in the end", "Hide"] ?

cspaier, 2019-11-22:
Thank you for the answer.

> If current behavior remains default, then I don't mind including it.

Of course current behavior would be default.

> How do you plan to implement it?

Don't know about technicals for now. I ll digg. Was just asking if you would accept such work before starting.

> Would it be easy to target #174 as well and introduce a three-valued configuration variable for imported names: ["No special treatment", "Show in the end", "Hide"] ?

Sounds like a very good plan.

aivarannamaa, 2019-11-22:
> Don't know about technicals for now.

I think the easiest option would be statically analyzing the AST of main module and collecting the names which are assigned to. The tricky part is differentiating between assignments to local vs global variables.

cspaier, 2019-12-02:
I am trying to understand how you get the variables informations at the moment.

In `plugins/variables` we receive an `event` that may contain a  `"globals"` key.
https://github.com/thonny/thonny/blob/09b0474c30a1271ffb203bd13a7621e3f6228f53/thonny/plugins/variables.py#L52

Then, after some checks, we pass this `event["globals"]` to `memory.MemoryFrame.update_variables()` method.

https://github.com/thonny/thonny/blob/09b0474c30a1271ffb203bd13a7621e3f6228f53/thonny/memory.py#L84

Sadly, I fail to understand where this `event["globals"]` is rendered in the first place.

Would you have any tip there @aivarannamaa ?

aivarannamaa, 2019-12-02:
Globals come from the backend process: https://github.com/thonny/thonny/blob/master/thonny/backend.py#L481

But I don't know how you could distinguish between imported and assigned names there. I think your best bet is to locate current file by `__file__`, parse it and find all names that have assignment in this module.

aivarannamaa, 2019-12-02:
You could do all this in the back-end, though. In this case I recommend adding another field (eg. `assigned_names`) to the result of `_cmd_get_globals` instead of extending the main structure for representing globals.

cspaier, 2019-12-02:
Thanks!

Will dig further.