thonny/thonny #999: writing characters over 7 bits makes two characters on disk

chrisdekort, 2019-11-22:
Hi, I am a newbie in Phyton Thonny an I already have a question.
when writing to a file I encounter following problem: (see print screen)
when writing a char over 127 like  chr(128) etc it writes 2 char to disk?
for instance chr(127) gives a 7F in the file but chr(128) gives a C2 80 in the file?
The 80 I understand but C2 80? What am I doing wrong?
[image]

aivarannamaa, 2019-11-22:
You are probably saving with UTF-8 encoding, and in this encoding only ASCII chars (ie. chars up to code 127) are written with one byte.

BTW. This is not a Thonny specific question.

closed 2019-11-22 by aivarannamaa

chrisdekort, 2019-11-22:
thanks for your quick response. Sorry for not knowing but how do I save the file correctly?

aivarannamaa, 2019-11-22:
> how do I save the file correctly?

Depends on what you want to achieve. I would say using utf-8 encoding is nowadays the correct way for storing text.

chrisdekort, 2019-11-22:
if I want to write up to 255 (FF) as one char to the file.

aivarannamaa, 2019-11-22:
Then you need to write bytes, not chars. For Python 2 these are the same but not for Python 3.

You need to open file in binary mode (`"bw"`) and use `bytes([...])` instead of `chr(...)`:

```
>>> chr(127)+chr(128)
'\x7f\x80'
>>> bytes([127])+bytes([128])
b'\x7f\x80'
```

aivarannamaa, 2019-11-22:
... the easiest is to collect all numbers to a list before converting to bytes:

```
>>> bytes([127, 128])
b'\x7f\x80'
```

chrisdekort, 2019-11-22:
Thank you so much 👍😀. life can sometimes be so simple with the right help!