thonny/thonny #1002: Inconsistent output when running ESP32 micropython code from the editor

effevee, 2019-11-23:
Hi, 

I'm using the latest Thonny 3.2.3 for programming MicroPython on ESP32 development boards. I noticed that running micropython code from the editor doesn't always produce correct output in the shell.

When I run this code to connect the ESP32 to my Wifi network from the editor :

```
import network
SSID = '*********'
PWD = '**********'

net = network.WLAN(network.STA_IF)
net.active(True)
net.connect(SSID, PWD)

print(net.isconnected())
print(net.ifconfig())

```
I get this output in the shell telling me that the wifi connection failed :

```

>>> %Run -c $EDITOR_CONTENT

I (34346) phy: phy_version: 4007, 9c6b43b, Jan 11 2019, 16:45:07, 0, 0
False
('0.0.0.0', '0.0.0.0', '0.0.0.0', '0.0.0.0')

```

However, if I give the same commands manually in the shell, I get different results saying that the ESP32 is in fact connected to my Wifi network.

```
>>> net.isconnected()
True
>>> net.ifconfig()
('192.168.1.18', '255.255.255.0', '192.168.1.1', '192.168.1.2')

```
This is all very strange. I'm running the latest micropython firmware on the ESP32

```
MicroPython v1.11 on 2019-05-29; ESP32 module with ESP32
Type "help()" for more information.
```
I have tested this with several different ESP32 dev boards with the same results, therefore I think that the problem lies in the Thonny editor.

BTW thanks for putting the firmware flashing back into Thonny; works perfectly.

Greetings,

Effevee

aivarannamaa, 2019-11-25:
I find it very unlikely that Thonny simply presents wrong output for `net.isconnected()` and `net.ifconfig()`, especially given that the result of these calls always match in your examples. It is more likely that sometimes your device connects to the WLAN and sometimes it doesn't.

I recommend asking around in https://forum.micropython.org/ about what can be the reason for such behavior.

If you are getting consistent behavior using the same script in another tool (eg. rshell), then I'm certainly willing to investigate further.

aivarannamaa, 2019-11-25:
I found the reason myself -- you are assuming `net.connect` is synchronous, but actually it is not. Sometimes the connection completes fast enough before you execute your print statements, sometimes print statements get executed before.

see the "useful function" example: https://docs.micropython.org/en/latest/esp32/quickref.html#networking

effevee, 2019-11-25:
You are right, I assumed `net.connect` would wait until the connection was established. When I think about it now, I see my error. In such a case the program would hang if there was something wrong with the wlan. Oops :-(

After adapting my code according to the example you suggested all is well. Thanks for your input :-)

closed 2019-11-25 by effevee