thonny/thonny #1003: Files window>rightclick >properties  wrong timestamp

xal88, 2019-11-23:
looks like you using wrong startdate for embedded devices, when calculating modified date/time
Missing 30 years ( 1.1.1970 instead of 1.1.2000 )
Example from ESP8266
[image]

aivarannamaa, 2019-11-24:
I'm just displaying what `os.stat()` reports for this file.

Are you saying it reports number of seconds from 1.1.2000 instead of Unix epoch time? Is there some documentation saying this?

aivarannamaa, 2019-11-24:
Found it already: https://docs.micropython.org/en/latest/library/utime.html
Thank you for bringing my attention to this! 

You're welcome to add more information if you have it!

xal88, 2019-11-24:
Hi,

dont have any more specific information than utime.html.

Thanks for the great program !

regards

Xal

 
 

Gesendet: Sonntag, 24. November 2019 um 15:41 Uhr

commit dbf3da09 (2019-12-06): FIXED #1003: Wrong interpretation of MicroPython file timestamps
closed 2019-12-06 by aivarannamaa