thonny/thonny #1007: On MicroPython device file update of sub-module fails

sengpw, 2019-11-26:
Not sure if this is a bug or just not implemented in Thonny 3.2.3.

"Project" consists of two files:
main.py and sub.py. Module sub.py is used by main.py.
When making changes to main.py and running program via icon or F5, program including changes is executed --> OK
When making changes to sub.py and running program main.py via icon or F5, program without changes is executed --> FAIL
When uploading sub.py to MicroPython device and running program main.py via icon or F5, program without changes is executed --> FAIL
When pressing "Refresh" on sub.py at MicroPython device and running program main.py via icon or F5, program without changes is executed --> FAIL
When pressing reset button at MicroPython device and running program main.py this way program with changes is executed --> OK

This behaviour is very error-prone. Is it possible to improve it?

sengpw, 2019-11-26:
When pressing CTRL-D (soft reboot) after uploading sub.py it has the same effect as pressing the reset button on the MicroPython device, program with changes is executed --> OK

aivarannamaa, 2019-11-26:
> When making changes to sub.py and running program main.py via icon or F5, program without changes is executed --> FAIL

This is probably acceptable, because Thonny only sends current script to the device.

> When uploading sub.py to MicroPython device and running program main.py via icon or F5, program without changes is executed --> FAIL

This is caused by (Micro)Python import machinery caching the previous version of module `sub`. Executing main again means simply sending next piece of code to MicroPython REPL, ie. executing `import sub` again reuses the cached module. This is different from CPython back-end, which restarts back-end Python process on each `%Run`.

I agree this is error-prone.

At the moment I don't know how to improve this (other than forcing a soft reboot on each run, which would be too heavy), but I posted a question in MicroPython forum: https://forum.micropython.org/viewtopic.php?f=2&t=7310

sengpw, 2019-11-27:
had a look at the replies to above link - can not rate whether some hints may work. But I think the solution should be obvious and simple, cause Thonny should not be error-prone to annoy first time users/learners.
What about an additional checkbox (normally enabled) in Tools-->Options-->General, like "Force soft reboot on each run"? So everything that is uploaded to the device will be used as current program code. Sure, can do this already now by pressing Ctrl-D, but I have to know it (and it must be explained)....
Right click on the editor file tab shows following options at the moment: "Close/Close others/Close "all. Extending the menu with "Upload to /"  or implementing an additional menu icon "Upload current script to /" would be very nice too.
Think intuitive Thonny can be really first time choice for MicroPython users/learners. It's really great already at the current  stage!

aivarannamaa, 2019-11-28:
I guess you're right -- automatic soft reboot would be good default choice. I didn't remember correctly how fast it is (provided there is nothing time consuming in boot.py).

I'm thinking of adding 3 options for what to do before each run:

* soft reboot (default)
* remove changed files from module cache (for cases where boot.py does something heavy, eg. connecting to WiFi)
* do nothing

I created another issue for "Upload current script" command: https://github.com/thonny/thonny/issues/1009

aivarannamaa, 2020-12-12:
Since version 3.3 Thonny does soft reboot on Run current script

closed 2020-12-12 by aivarannamaa