thonny/thonny #1008: opencv import cv2 error

max54654654, 2019-11-28:
I install Opencv successfully, and can run gui_python_test.py in LX terminal, but when I run gui_python_test.py in Thonny Python IDE, it shows "ImportError: libopencv_quality.so.4.1: can not open shared object file: No such file or directory",   What can I do? many thanks!

aivarannamaa, 2019-11-28:
You should first check whether you can run your script from command line without Thonny. If yes, then I'm willing to dig further. If still not, then you should google or ask in OpenCV forums.

ozgunyargi, 2020-03-01:
Let’s dig further

aivarannamaa, 2020-03-02:
How did you install Opencv? Can you post a minimal script that demonstrates your problem?

Gitboeff, 2020-09-06:
I have exactly the same Problem.
I installed open CV on this tutorial:
https://solarianprogrammer.com/2019/09/17/install-opencv-raspberry-pi-raspbian-cpp-python-development/
and it works.
the test Programm:

import cv2
print("OpenCV version:")
print(cv2.__version__)
img = cv2.imread("lake.jpg")
gray = cv2.cvtColor(img, cv2.COLOR_BGR2GRAY)
cv2.imshow("Lake", img)
cv2.imshow("Lake - gray", gray)
cv2.waitKey(0)
cv2.destroyAllWindows()

works, if I start it from the Terminal with 
python3 gui_python_test.py
(I get two pics, one in Color, one in gray)

but if I start the same with Thonny, I get this Error:

File "/home/pi/gui_python_test.py", line 2, in <module>
    import cv2
File "/usr/lib/python3/dist-packages/cv2/__init__.py", line 90, in <module>
    bootstrap()
File "/usr/lib/python3/dist-packages/cv2/__init__.py", line 80, in bootstrap
    import cv2
ImportError: libopencv_quality.so.4.1: cannot open shared object file: No such file or directory

I found the following files:

libopencv-quality.so in the following directories:
/home/pi/opencv-4-1-0/lib
/opt/opencv-4.1.0/lib

libopencv-quality.so.4.1 in the following dirctories:
/home/pi/opencv-4-1-0/lib
/opt/opencv-4.1.0/lib

libopencv-quality.so.4.1.0 in the following dirctories:
/home/pi/opencv-4-1-0/lib
/opt/opencv-4.1.0/lib

I think, i have to copy something to somewhere, but I don't know "what to where".

Perhaps i can get some help...

Stefan

aivarannamaa, 2020-09-07:
@Gitboeff, are you on Raspberry Pi and using Thonny as it came with it?

If you had installed Thonny+Python binary bundle, I would propose to unify LD_LIBRARY_PATH described in the tutorial with LD_LIBRARY_PATH in Thonny's launcher.

Gitboeff, 2020-09-07:
Thanks, now the problem is solved.

The tutorial told me to make the following symbolic link:

sudo ln -s /opt/opencv-4.1.0/lib/python3.7/dist-packages/cv2 /usr/lib/python3/dist-packages/cv2

I add two different links:
sudo ln -s /opt/opencv-4.1.0/lib/python3.7/dist-packages/cv2 /usr/lib/python3/dist-packages/thonny/cv2
and
sudo ln -s /opt/opencv-4.1.0/lib/python3.7/dist-packages/cv2 /usr/lib/python3/dist-packages/thonny/plugins/cv2

and now it works.
I don't know which one of the two was "the one that helps". this don't matter anymore.

Thanks for your help.

closed 2020-09-20 by aivarannamaa

Petros626, 2023-02-28:
@Gitboeff could you tell me why the tab completion isn't working? I installed `opencv-python `via **Manage packages** in the IDE.