thonny/thonny #1011: Include Jupyterlab in XXL

aivarannamaa, 2019-12-02:
... together with a launcher menu item

eb235, 2021-12-08:
I'm not sure this is the right place to post the issue I'm having.

When I tried to create a virtual environment in Thonny with jupyter package, I got an error saying rust and cargo were required.
I installed rust-autoloader (the package called RUST is something totally different). I checked that I did still need cargo by trying to install Jupyter again. Then I installed cargo-lite only to get a long-winded error message.
(All my installs were through Thonny's package manager)

The real problem/question is : how come the same process of installing jupyter in a virtual environment in PyCharm (on the same Windows 10 computer) goes without a hitch and **without** installing anything related to rust or cargo ???
(base Python interpreter 3.7 for Thonny, 3.9 for PyCharm - it seems Thonny can't use an alternative interpreter for creating a virtual environment)

aivarannamaa, 2021-12-09:
@eb235, it may be that one of the jupyter dependecies has published pre-built wheels for Python 3.9, but not for 3.7 (which is currently falling out of support for many packages)

rootsmusic, 2023-11-05:
(@aivarannamaa  Jupyter notebook is very helpful in my learning.  By being able to run only the cells that I choose or to re-run one cell after modifying its content, I can debug by cell without a debugger.)  Since I didn't find a Wiki about Jupyter, I'm assuming that .ipynb files still aren't supported by Thonny.