thonny/thonny #1012: split screen creates offset between caret and mouse pointer

jmsoeiro, 2019-12-02:
When splitting a screen between thonny and other application, the mouse pointer and the caret  / insertion pointer / button hotspot are vertically offset
Mouse pointer is roughly a line below the caret and the hotspot area below the button area. 

This is mostly troublesome when selecting  very ragged texts  ( code ....) .

The images show the difference of tooltip placement, the first correctly by the button, the second wrongly away, probably next to some *shadow box*.

Also, when the other application quits split screen, thonny is maximized and it seems that the only way to exit full screen is to close ^Q thonny.
[image]
[image]

## Verified on
MAC OSX Catalina 
  System Version:	macOS 10.15.1 (19B88)
  Kernel Version:	Darwin 19.0.0
MAC OSX Mojave
  System Version:	macOS 10.14.6 (18G103)
  Kernel Version:	Darwin 18.7.0

aivarannamaa, 2019-12-02:
Thank you for the report! I'll try to reproduce it later.

aivarannamaa, 2019-12-03:
I was able to reproduce it, but it looks like it's a problem on the framework (Tk) level. For example Python's IDLE has same problem. There is hope that this will be fixed with next version of Python 3.7 / 3.8 (provided they also include newest version of Tk): https://wiki.tcl-lang.org/page/Tk+differences+on+Mac+OS+X

aivarannamaa, 2021-02-27:
Fixed by upgrading to Tk 8.6.11. 

Experimental build available in https://github.com/thonny/thonny/releases/download/v3.3.5/thonny-3.3.5_alt.pkg (bundled with Universal 2 binaries of Python 3.9.2)

closed 2021-02-27 by aivarannamaa