thonny/thonny #1013: Error When I try to open a file from micropython device (NodeMCU esp8266)

AHardReset, 2019-12-04:
Instead of show the files in the UI, it reports in repl.

[image]

This happens after first erase flash, so if I erase the flash again and upload the .bin file then Thonny wors as expected
 
Next day I try to work the issue comes again

aivarannamaa, 2019-12-04:
I don't understand -- how did you get to MicroPython REPL after erasing the flash?

AHardReset, 2019-12-10:
1.- Erase flash and after write-flash with the .bin file from micropython webpage
2.- Everything works as expected
3.- In future sesions I reach this problem

aivarannamaa, 2019-12-10:
Does resetting and reconnecting your device also cure this problem?

RobertBlackhart, 2020-02-16:
I've seen this problem happen many times in the past few days that I've been using Thonny. It seems like it is worse the more files there are to list. That is, it happens more often if I have more files stored on the device.

In order to get out of this problem, I need to cancel the dialog window, then soft-reset my device once or twice so that the ">>>" appears in the REPL. Then I can try again. It may or may not work and I might need to repeat these steps several times.

aivarannamaa, 2020-02-16:
Thank you for the information! I'll try to reproduce it.

RobertBlackhart, 2020-02-16:
Might be helpful too. Sometimes I see the files listed in the window and have the text in the REPL at the same time:

[image]

In this case, I cannot open a file. If I try, it will fail (hangs forever in the open progress dialog).

antrapoint, 2020-02-22:
I got the same issue if i had subdirs on my micropython device. 
My workaround for Thonny 3.2.6 32bit :
TOOLS -> Open Thonny programm folder ...
Open Thonny/lib/queue.py
Goto line 153 ("_**def get(self, block=True, timeout=None**_):"
Add under the docstring: **timeout = 4**
Save and restart Thonny

This works fine for me, maybe it's the same issue. Hope it help!

aivarannamaa, 2020-10-28:
Should be fixed in 3.3 (beta)

closed 2020-10-28 by aivarannamaa