thonny/thonny #1014: __file__ is not defined

bendavis78, 2019-12-05:
When running a simple python script, `__file__` cannot be used.

    # test.py
    print(__file__)

When run the script, you get the error:

    Traceback (most recent call last):
      File "/home/me/code/test.py", line 1, in <module>
        print(__file__)
    NameError: name '__file__' is not defined

aivarannamaa, 2019-12-06:
I can't reproduce this. Which Thonny version on which platform are you using?

ssermus, 2019-12-07:
I also can't reproduce this error, at least not on Windows 10 64-bit. It seems he uses it on linux (from the file structure).

bendavis78, 2019-12-27:
This was in Arch Linux, however it doesn't seem to be an issue on Ubuntu. Could just be a botched install. Closing for now.

closed 2019-12-27 by bendavis78

Donatussss, 2024-02-19:
Hi, I'm getting this issue. I'm using
- Thonny 4.1.4
- Windows 10 Pro
- Version 22H2
- Installed on 15/‎04/‎2021
- OS build 19045.2965

Any help regarding the same will be appreciated