thonny/thonny #1016: install of thonny-black-format does not work

h1moodle, 2019-12-06:
Hello from Germany,

i install the plugin and all seems to be ok, but if i try to use the plugin via STRG-ALT-C i se he error message, "could not find black package. I it installed in PATH?

IN thonny i se the installation path for the plugin: C:\Users\Heinz\AppData\Roaming\Python\Python37\site-packages

in this i see the plugins black-19.10b0.dist-info and thonny_black_format-0.0.1a2.dist-info

any ideas what´s wrong with the installation?

Cheers 

Heinz

aivarannamaa, 2019-12-07:
Hi!

This looks like a problem in the plug-in. I created an issue for it: https://github.com/Franccisco/thonny-black-code-format/issues/1

closed 2019-12-07 by aivarannamaa

Franccisco, 2019-12-07:
Hello! I just published a new package version in PyPI. Please, upgrade your current thonny-black-format package and tell me if that solution works for you.

h1moodle, 2019-12-07:
Hello Francisco,

i installed our update an see a different result:

[image: image.png]

If i edit the line to be correct the plugin shows:

[image: image.png]

I thought that the plugin will do the reformatting?

Greetings from Blackforest

Heinz

Franccisco, 2019-12-07:
Sorry, I can't see the images you uploaded. Can you attach them directly from github?

h1moodle, 2019-12-07:
[image]

[image]

Franccisco, 2019-12-07:
That means the plugin is working. In summary:

- In the first image, the Error! popup lets you know that there's an error in your code. While you don't fix it, the plugin won't be able to apply the Black format.
- The second image is telling you that everything went fine and your code is now formatted with Black.

Thank you guys for reporting the issue and helping!

h1moodle, 2019-12-07:
Thxs for info :-)

cybarspace, 2020-12-10:
I know the issue is closed, but I'm having some trouble with getting Black to work in Thonny. If I do `Tools > Format with Black`, that works fine. If I press `Ctrl + Alt + C`, it doesn't show the popup but instead shows a `KeyboardInterrupt` line in the Shell.
Is there any way to fix the keyboard shortcut or remap the binding to something that doesn't include `Ctrl` and `C` (the default Keyboard interrupt hotkey for most terminals)?

PS: I am using Manjaro Linux XFCE, Zsh login shell, Python 3.90, Thonny 3.3.1 from AUR.

[image]

aivarannamaa, 2020-12-10:
@cybarspace, I see you have already modified configuration.ini by hand. Changing a shortcut is described here: https://github.com/thonny/thonny/wiki/Custom-shortcuts

cybarspace, 2020-12-25:
> I see you have already modified configuration.ini by hand.

Yes, I prefer the shell to be split vertically. :sweat_smile:
> Changing a shortcut is described here: https://github.com/thonny/thonny/wiki/Custom-shortcuts

Done. Now I've remapped it to a more convenient shortcut. Thanks a bunch. :grinning:

wrightkevin, 2022-11-28:
Get this problem with Thonny 4.0.1. Error message is - Could not find Black package. Is it installed on on your PATH? 

Please advise.

aivarannamaa, 2022-12-28:
@wrightkevin, I assume you are having problem with the black formatter Thonny plug-in. In this case please report your problem to the developer of the plug-in.