thonny/thonny #1017: Backend Error on startup

thamilton533, 2019-12-07:
I have been using a Thonny for a few weeks now and last night I started getting the error below.

This error occurs when the program starts, not when I run script.

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/thonny/backend_launcher.py", line 76, in <module>
    VM().mainloop()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 122, in __init__
    self._load_shared_modules()
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 298, in _load_shared_modules
    self.load_modules_with_frontend_path(["parso", "jedi", "thonnycontrib", "six", "asttokens"])
  File "/usr/lib/python3/dist-packages/thonny/backend.py", line 308, in load_modules_with_frontend_path
    import_module(name)
  File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/lib/python3/dist-packages/jedi/__init__.py", line 41, in <module>
    from jedi.api import Script, Interpreter, set_debug_function, \
  File "/usr/lib/python3/dist-packages/jedi/api/__init__.py", line 29, in <module>
    from jedi.api.project import get_default_project
  File "/usr/lib/python3/dist-packages/jedi/api/project.py", line 2, in <module>
    import json
  File "/usr/lib/python3/dist-packages/flask/json/__init__.py", line 13, in <module>
    from flask.globals import current_app, request
  File "/usr/lib/python3/dist-packages/flask/__init__.py", line 19, in <module>
    from jinja2 import Markup, escape
  File "/usr/lib/python3/dist-packages/jinja2/__init__.py", line 33, in <module>
    from jinja2.environment import Environment, Template
  File "/usr/lib/python3/dist-packages/jinja2/environment.py", line 15, in <module>
    from jinja2 import nodes
  File "/usr/lib/python3/dist-packages/jinja2/nodes.py", line 19, in <module>
    from jinja2.utils import Markup
  File "/usr/lib/python3/dist-packages/jinja2/utils.py", line 41, in <module>
    _slash_escape = '\\/' not in json.dumps('/')
AttributeError: module 'json' has no attribute 'dumps'

Backend terminated or disconnected. Use 'Stop/Restart' to restart ...

aivarannamaa, 2019-12-07:
Do you happen to have `json.py` in your working directory? If yes, then you should rename it because its name clashes with a standard library module name.

thamilton533, 2019-12-07:
Hi Aivar,

I am not sure that I know how to check that. Could you give me some
pointers? I have tried re-installing Thonny and that did not work.

Thanks,
Tim

aivarannamaa, 2019-12-07:
* In Thonny select "View => Files" (unless you already have filebrowser visible)
* In left side of Thonny's window, there should be now list of directories and files. 
* Do you see file named `json.py` among these?

aivarannamaa, 2019-12-07:
In addition to previous comment -- which directory gets opened in the filebrowser?

In case you don't see json.py, please try following:

* Open terminal from Thonny (Tools => Open system shell) 
* Execute following command: `echo $PYTHONPATH`
* What do you get?

thamilton533, 2019-12-07:
I see a json folder in the file directory. Won't renaming break the any
where else it is used?

[image: image.png]

aivarannamaa, 2019-12-07:
GitHub doesn't show email attachments properly, so I don't see your shortcut. Please post it in GitHub interface or tell me which directory is opened in Thonny's filebrowser.

thamilton533, 2019-12-07:
Apologies, I have attached the image. 

[image]

aivarannamaa, 2019-12-07:
Each process running in computer has a *current directory* or *working directory*. For Python this is important because this is the first directory where it tries to resolve `import`-s.

Thonny assigns working directory to Python process according to which directory is open in its filebrowser or which directory the script to be run is stored in.

You have made `/usr/lib/python3/dist-packages/flask` to be working directory for Python (either by navigating into this directory in filebrowser or by running a script from it). For this reason Python can't find a standard library module called `json` because the `json` in working directory takes precedence.

The solution is to navigate away from this directory and press red "Stop/restart" button.

thamilton533, 2019-12-07:
Ahh, I understand. I have resolved the issue. 

Thank you for your help,
Tim

closed 2019-12-07 by thamilton533

diamonta222, 2023-05-18:
[image]
thonny won t start on windows 7 64 bits i have this message

aivarannamaa, 2023-05-25:
@diamonta222, which Thonny version is this? Thonny 4.0.2 with 32-bit Python 3.8?