thonny/thonny #1019: AttributeError: 'NoneType' object has no attribute 'is_waiting_toplevel_command'

jedie, 2019-12-07:
```
Traceback (most recent call last):
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/__init__.py", line 166, in launch
    bench.mainloop()
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1283, in mainloop
    self.tk.mainloop(n)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/tktextext.py", line 45, in _dispatch_tk_operation
    return f(*args)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/shell.py", line 821, in intercept_insert
    if get_runner().is_waiting_toplevel_command():
AttributeError: 'NoneType' object has no attribute 'is_waiting_toplevel_command'
ERROR: Internal launch or mainloop error
Traceback (most recent call last):
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/__init__.py", line 166, in launch
    bench.mainloop()
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1283, in mainloop
    self.tk.mainloop(n)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/tktextext.py", line 45, in _dispatch_tk_operation
    return f(*args)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/shell.py", line 821, in intercept_insert
    if get_runner().is_waiting_toplevel_command():
AttributeError: 'NoneType' object has no attribute 'is_waiting_toplevel_command'
```

aivarannamaa, 2019-12-09:
I guess I need more information. Which action produced this stracktrace? Which Thonny version?

jedie, 2019-12-09:
It the current 3.2.1 and it happens sometimes directly after start. But only sometimes.

aivarannamaa, 2019-12-20:
I was able to get same stacktrace when I delayed starting of the backend and typed into shell before the backend was ready. This case should now be fixed.

jedie, 2019-12-20:
Sounds good !

closed 2019-12-20 by aivarannamaa