thonny/thonny #1023: Autocomplete error 3.2.4 MicroPython

cmdrdeliver, 2019-12-09:
Thonny 3.2.4 using Interpreter MicroPython 8266

Steps to reproduce:
- New File
- Save (junk.py)
- Type "imp" hit tab

Result: 
Dialog "Autocomplete Error" : "Autocomplete Error"

aivarannamaa, 2019-12-12:
I can't reproduce this in Windows with Thonny 3.2.4.

Did you save the file locally or to the board?

Can you please post the stacktrace?

cmdrdeliver, 2019-12-16:
Aivar,
I've attached a video showing process.  Not getting a stack trace.  Only occurs when the board is connected.  Had to zip video for supported attachment types.

Mike
[ThonnyAutocomplete3.2.4..mp4.zip](https://github.com/thonny/thonny/files/3968198/ThonnyAutocomplete3.2.4.mp4.zip)

aivarannamaa, 2019-12-16:
Thanks! I was able to reproduce this in Linux

closed 2019-12-16 by aivarannamaa