thonny/thonny #1026: Base64 images cannot contain "+"

acbart, 2019-12-16:
I'm like 70% sure this what's happening. The Regex here:

https://github.com/thonny/thonny/blob/af28058024e4d53aed4cd5bb46ad1330760ffd9b/thonny/shell.py#L36

Does not allow data URLs that contain "+". But that's a [valid character for base64 data](https://base64.guru/learn/base64-characters).

aivarannamaa, 2019-12-16:
I think I copied the regex from somewhere. Can you fix this?

acbart, 2019-12-16:
Yep, the change is trivial, just the addition of an escaped `+` near the end.

```
SIMPLE_URL_SPLIT_REGEX = re.compile(r"(https?:\/\/[\w\/.:\-\?#=%]+[\w\/]|data:image\/[a-z]+;base64,[A-Za-z0-9\/=\+]+)")
```

I can make a pull request at some point, or you could just drop in the two extra characters

aivarannamaa, 2019-12-20:
Seems to be fixed now

closed 2019-12-20 by aivarannamaa