thonny/thonny #1029: cannot import name 'execute_system_command' from 'thonny.common'

ShenTengTu, 2019-12-20:
I have synced my forked repository today. 

I start up thonny and use `Micropython(ESP32)`  backend, then get the error as below:
```
Traceback (most recent call last):
  File "d:\MainWorkSpace\WORKPROJECT\CONTRIBUTING\thonny\thonny\plugins\micropython\backend.py", line 1, in <module>
    from thonny.common import (
ImportError: cannot import name 'execute_system_command' from 'thonny.common' (D:\Python37\lib\site-packages\thonny\common.py)
```

aivarannamaa, 2019-12-20:
The last line shows, that Thonny's back-end imports Thonny from `D:\Python37\lib\site-packages\thonny` instead of `d:\MainWorkSpace\WORKPROJECT\CONTRIBUTING\thonny`. You should run your development Thonny with `PYTHONPATH` environment variable set to `d:\MainWorkSpace\WORKPROJECT\CONTRIBUTING\thonny`. I updated documentation accordingly: https://github.com/thonny/thonny/wiki/Running-from-the-source

ShenTengTu, 2019-12-23:
It seems that I did not remove the installed Thonny module. Thank reply.

closed 2019-12-23 by ShenTengTu