thonny/thonny #1035: AttributeError: 'IfStmt' object has no attribute 'get_statement_for_position'

jedie, 2019-12-23:
Thonny v3.2.1 direct after start:
```
Traceback (most recent call last):
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/__init__.py", line 166, in launch
    bench.mainloop()
  File "/usr/lib/python3.6/tkinter/__init__.py", line 1283, in mainloop
    self.tk.mainloop(n)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/tktextext.py", line 45, in _dispatch_tk_operation
    return f(*args)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/shell.py", line 821, in intercept_insert
    if get_runner().is_waiting_toplevel_command():
AttributeError: 'NoneType' object has no attribute 'is_waiting_toplevel_command'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/jedi_utils.py", line 45, in get_statement_of_position
    from jedi.parser_utils import get_statement_of_position
ImportError: cannot import name 'get_statement_of_position'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/plugins/highlight_names.py", line 98, in _get_statement_for_position
    return jedi_utils.get_statement_of_position(c, pos)
  File "/home/jens/.local/share/virtualenvs/thonny-6qUXfC5h/lib/python3.6/site-packages/thonny/jedi_utils.py", line 50, in get_statement_of_position
    return node.get_statement_for_position(pos)
AttributeError: 'IfStmt' object has no attribute 'get_statement_for_position'
```

Sometimes it crashed completely: GUI closed. But not every time.
I have some files open and my ESP8266 is not connected at startup.

aivarannamaa, 2019-12-25:
It looks like duplicate of #1019

I'll release 3.2.5 in couple of days. Hopefully this gets fixed then.

jedie, 2019-12-27:
Oh. pipenv stayed with v3.2.1 because i have also 'thonny-esp' in my Pipfile... Now i see on https://pypi.org/project/thonny-esp/ that's included in thonny...

Maybe it's a good idea to change "thonny-esp" to a "empty" dummy package or something else?!?

aivarannamaa, 2019-12-30:
Can you please check the versions of your `jedi` and `parseo` packages:

```
>>> import jedi
>>> jedi.__version__
'0.15.2'
>>> import parso
>>> parso.__version__
'0.5.2'
```

jedie, 2019-12-30:
I have the same versions:

```
>>> import jedi
>>> jedi.__version__
'0.15.2'
>>> import parso
>>> parso.__version__
'0.5.2'
```

I use thonny in a "empty" virtualenv via pipenv:
```
$ cat Pipfile
[[source]]
name = "pypi"
url = "https://pypi.org/simple"
verify_ssl = true

[dev-packages]

[packages]
thonny = "*"

[requires]
python_version = "3.6"
```

aivarannamaa, 2020-01-01:
Should be fixed now

aivarannamaa, 2020-01-05:
Hmm, name highlighting still has problems.