thonny/thonny #1036: Horizontal scrollbar not automatically showing in editor

eb235, 2019-12-23:
I noticed that you had to click on a line longer than the current width of the editor and scroll it manually (i.e. move the cursor) to make the horizontal scrollbar appear below the editor.
To me it would be better to have the horizontal scrollbar appear as soon as a line in the editor has a width larger than the width currently showing in editor.
The reason being that one may want to check what's written in another (long) line without moving the cursor away from the line where it's sitting.

FYI I'm using Thonny 3.2.1 with recommended max line length set to 0 (another value has no effect) on Windows 8.1 with portrait orientation.
Thank you

aivarannamaa, 2019-12-25:
Thank you for the report!

This seems to be a peculiarity of Tk's scrollbars. In next Tk version (8.6.10) there are some fixes related to scrollbar. I'll postpone this issue until Python starts building with 8.6.10.

Related to #659

lurch, 2020-05-31:
IMHO this should be closed as a duplicate of #659 
I've investigated this problem and found it's actually a problem in Thonny itself rather then Tk. I've added a big comment in #659 with the first steps towards a solution... :slightly_smiling_face:

aivarannamaa, 2020-07-24:
Duplicate of #659

closed 2020-07-24 by aivarannamaa