thonny/thonny #1038: Autocomple not working with pygame

bendavis78, 2019-12-27:
I'm not sure if this is an issue with Thonny or with some other component. But, when I type the following:

    import pygame
    pygame.display.

and hit <kbd>Tab</kbd>, I expect it to show the contents of the `display` module, which includes methods such as `.get_mode()`, however this is what I get:

[image]

aivarannamaa, 2019-12-30:
It looks like the autocompletion library (jedi) can't handle some tricks Pygame plays in this module. Here is a similar issue: https://github.com/davidhalter/jedi-vim/issues/146

It may get fixed in newer Jedi or Pygame versions, but most likely I can't do anything about it in Thonny.

closed 2019-12-30 by aivarannamaa

bendavis78, 2020-01-03:
Indeed this looks like an issue pygame needs to fix. It looks like it was fixed once before, but pygame had a major overhaul recently and the bug regressed. This issue seems to be the one to follow in case anyone else is interested: https://github.com/pygame/pygame/issues/1267

Seraphli, 2020-01-22:
I searched for this issue on Google and found out the reason might be the `try-except` expression in the `pygame/__init__.py`.