thonny/thonny #1042: thonny throws exception when saving file

kr-g, 2019-12-28:
details:

Traceback (most recent call last):
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/workbench.py", line 1598, in event_generate
    handler(event)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 790, in on_local_file_operation
    self.refresh_tree()
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 640, in refresh_tree
    self.render_children_from_cache("")
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 382, in render_children_from_cache
    self.request_dirs_child_data(node_id, self.get_open_paths() | {path})
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 765, in request_dirs_child_data
    self.render_children_from_cache(node_id)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 444, in render_children_from_cache
    self.render_children_from_cache(child_id)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 382, in render_children_from_cache
    self.request_dirs_child_data(node_id, self.get_open_paths() | {path})
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 765, in request_dirs_child_data
    self.render_children_from_cache(node_id)
  File "/home/benutzer/.local/lib/python3.6/site-packages/thonny/base_file_browser.py", line 402, in render_children_from_cache
    raise RuntimeError("None data for %s" % path)
RuntimeError: None data for /home/benutzer/repo/pyfileguard/monty.txt

aivarannamaa, 2019-12-30:
Thank you for the report!

Does it happen when you save a file in your program or when saving Python script with Thonny?

Can you give me exact steps how to reproduce this?

kr-g, 2019-12-30:
during save of a file in thonny.
happened one evening, also after restarting thonny it appeared again.
i also can _not_ reproduce it now.

closed 2021-01-19 by aivarannamaa