thonny/thonny #1045: Total Python noob

WarningOldFart, 2019-12-30:
A reminder, I am a complete noob.
Do not berate me.
If you do not have anything nice to say, don't say anything!
Please, be patient with me!
I haven't done anything like this in years!

I tried to run the Hello World code in Thonny (on Lubuntu, which I am also a noob at).

When I try to run it I wants me to save to a folder.

I select a folder and it will not let me save it!
I tried selecting other formats from the dropdown menu (Text, All files) to save.
The OK button doesn't highlight!

What gives?

Did I skip a step or is there something wrong with this version of Thonny (3.1.2) ?
Or is it me?

Thank you.

WarningOldFart, 2019-12-30:
Never mind!

I think I got it!

I forgot to give the Hello World code a title when I attempted to save it.
(Damn! I got a lot of learning to do!)

Thanks anyway.
Sorry to bother you.

If I have any other issues, hopefully they will be a lot more serious.

closed 2019-12-30 by WarningOldFart

aivarannamaa, 2019-12-31:
No problem! Happy programming!