thonny/thonny #1048: Thonny Configuration Issue

eggehad, 2020-01-02:
Thonny 3.1.2 on Fedora Workstation 31, 
I've run into an issue everytime I try to spawn an external command in python (under thonny):

os.system(runstring)
/usr/bin/HandBrakeCLI: /home/david/apps/thonny/bin/../lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /lib64/libssh.so.4)
/usr/bin/HandBrakeCLI: /home/david/apps/thonny/bin/../lib/libcrypto.so.1.1: version `OPENSSL_1_1_1b' not found (required by /lib64/libssh.so.4)
or
os.open(runstring)
/usr/bin/HandBrakeCLI: /home/david/apps/thonny/bin/../lib/libcrypto.so.1.1: version `OPENSSL_1_1_1' not found (required by /lib64/libssh.so.4)
/usr/bin/HandBrakeCLI: /home/david/apps/thonny/bin/../lib/libcrypto.so.1.1: version `OPENSSL_1_1_1b' not found (required by /lib64/libssh.so.4)

if I run  openSSL directly, I'm good:
[david@nas2 lib]$ export LD_LIBRARY_PATH=/usr/lib64
[david@nas2 lib]$ openssl version
OpenSSL 1.1.1d FIPS  10 Sep 2019

If I run HandBrakeCLI directly I'm also good:
[david@nas2 lib]$ HandBrakeCLI --version
[08:31:51] Nvenc version 9.1

[08:31:51] hb_init: starting libhb thread
[08:31:51] thread 7f07a8e28700 started ("libhb")
HandBrake 20191209131810-a5d359d-unknown

HandBrake has exited.
[david@nas2 lib]$ 

and more importantly, when I run the script from the command line, I'm good too!
It only breaks consistently when running under Thonny!
Not sure how to proceed to fix this.

Any suggestions?

-d

aivarannamaa, 2020-01-02:
Are you using Thonny's binary bundle (Python+Thonny)? This variant sets its own `LD_LIBRARY_PATH`: https://github.com/thonny/thonny/blob/master/packaging/linux/thonny

eggehad, 2020-01-26:
Thank you, this was the issue.

lurch, 2020-05-23:
Can this be closed then?

aivarannamaa, 2020-05-23:
> Can this be closed then?

I'll leave it open for now, as a reminder to pay attention to the environment when launching subprocesses from Thonny.

eggehad, 2020-05-24:
yes, it can be closed.  Your comments helped me work around the problem.
thank you!

-david

aivarannamaa, 2023-08-19:
Closing as starting with Thonny 5.0 I won't be publishing Linux binaries anymore, only the installation script based on system Python.

closed 2023-08-19 by aivarannamaa