thonny/thonny #1049: Implement HTML editor

aivarannamaa, 2020-01-05:


lurch, 2020-05-23:
Ugh, really? Wouldn't it be better for Thonny to focus on Python and let other editors focus on HTML?

aivarannamaa, 2020-05-23:
What I had in mind was a really simple editor for students who learn web programming with Flask. Just coloring the tags and highlighting mismatched start and end tags (quite like unbalanced parens in Python).

It's definitely low priority and may not get done.

SoproLab, 2020-05-23:
Hello,

It will be very very useful if it was possible to edit .html .css and .js files !!!

The problem we have currently in France is that we have to study IOT with students between 15 - 18 years old.

I use Thonny several hours every day because I think it is a very good IDE for micro python with ESP32.
So, for IOT I use the Jean-Christophe BOS microweb server with my students but … but we can’t edit .html .css and .js files.

At the moment, I ask my students to write the code with visual studio code, save the file / download the file with Thonny …
When they have only to change some words, some code we spend too much time for that.

What you propose : «  What I had in mind was a really simple editor for students who learn web programming with Flask. Just coloring the tags and highlighting mismatched start and end tags (quite like unbalanced parens in Python). » will be very very useful !

The notice I use with My students :
https://github.com/SoproLab/Soprolab/blob/master/Notice_Utilisation.pdf <https://github.com/SoproLab/Soprolab/blob/master/Notice_Utilisation.pdf>
Sorry it is in French but You will see that Thonny is used in every pages and also in the others videos : https://j-chouteau.org/carte-soprolab <https://j-chouteau.org/carte-soprolab>

Thank you very very much for not having forgotten my message.  ;-)

See you soon

Jacques

SoproLab, 2020-05-27:
Hello,
I come back to my previous message.
If it is not possible to create a syntax editor for html / css / javascript files, I understand completely since it represents a lot of work, would it be possible simply to be able to open that kind of files in the editor as a txt or csv file?
It would already be super practical ;-)

Thank you very much.
Jacques

aivarannamaa, 2020-10-18:
Plain text html editor will be in Thonny 3.3 (present in current beta).

SoproLab, 2020-10-18:
For a GOOD news is e very good news ;)

Thank you a lot.

Last week I had 4 days working with other professeurs. The solution we found was to write the html in a text file -> html.txt but it was a solution not very useful for explain that at students …

So I will try to use it tomorrow !!!!

Thank you l a lot

fgallaire, 2026-02-14:
> What I had in mind was a really simple editor for students who learn web programming with Flask. Just coloring the tags and highlighting mismatched start and end tags (quite like unbalanced parens in Python).

Hello, professor here, exactly the same usecase

> It's definitely low priority and may not get done.

High priority IMH!

fgallaire, 2026-02-14:
HTML syntax highlighting is the most important, but JS and CSS could be occasionally useful.

There's great libs doing that stuff (https://pygments.org)  so it should be easy to add.