thonny/thonny #1050: portable mode dose not work for me

haha198566, 2020-01-05:
I download the last protable version of thonny(thonny-3.2.6-windows-portable.zip),exact it to d:\thonny,launch thonny.exe,it create folder in appdata/roaming and appdata/local. There's no user_data folder which is created in the main folder of program, I try to make the "user_data" folder in d:/thonny and D:\thonny\Lib\site-packages\thonny,but it always put .log and configration files into %appdata%!is anything wrong?

aivarannamaa, 2020-01-05:
Thonny currently switches to portable mode only when it is run from a removable drive (eg. USB stick).

I may add another options in a future version

haha198566, 2020-01-06:
> Thonny currently switches to portable mode only when it is run from a removable drive (eg. USB stick).
> 
> I may add another options in a future version

It makes sense now ,thank you .

haha198566, 2020-01-06:
I figured it out,it worked for me.
1 rename thonny_python.ini to portable_thonny.ini(where the folder with thonny.exe in it)
2 edit thonny\lib\site-packages\thonny\__init__.py :
def _get_ipc_file_path():
     from thonny import misc_utils     import getpass
     if  is_portable() and platform.system() == "Windows":
         base_dir = THONNY_USER_DIR
     elif platform.system() == "Windows": 
        base_dir = misc_utils.get_local_appdata_dir()
     ...
3 save it
!!!! It just work for me not others.
(not a native English speaker,so...)

stdedos, 2020-01-14:
Duplicate of https://github.com/thonny/thonny/issues/697, and already resolved?

However, I think the shorthand https://github.com/thonny/thonny/blob/master/thonny/__init__.py#L48 :
i) Should be the default in the portable artifacts (i.e. touch the file in https://github.com/thonny/thonny/commit/eee3db406052b70f0edcca4a045f34491e436202 aka while creating `thonny-x-windows-portable.zip` artifact)
ii) Should be mentioned in https://github.com/thonny/thonny/wiki/DeploymentOptions#portable-version

aivarannamaa, 2020-01-14:
@haha198566, do I understand correctly, that with your edit in point number 2 you want to allow simultaneously use portable and installed Thonny instances?

@stdedos, you're right. I had forgotten about "portable_thonny.ini". Thank you for reminding me this! :)