thonny/thonny #1051: Remove [BETA] from Greek Language

niplessa, 2020-01-05:
I have reviewed my Thonny translation and corrected some mistakes. It now is error - free (as possible). 
Please I would like the [Beta] Tag to be removed.

Thank you.

aivarannamaa, 2020-01-07:
Will do. Thanks for your work!

Hermann12, 2020-01-09:
I saw the German version is a mix-up English/German. How can I help? I am native German.

aivarannamaa, 2020-01-09:
@Hermann12 , you are welcome to join translators here: https://poeditor.com/join/project/Gh188fdYH6 and add German translations to missing items

Hermann12, 2020-01-10:
Done. See my post on German Update. Maybe this item is still closed sorry. I have tested as described. It doesn^t show all translations. In the POEditor I cańt find e.g. Run the script into the .mo file. Are not all menu labels for translation? How can I send you the .mo and .po files? Thanks.

aivarannamaa, 2020-01-10:
@Hermann12, I made new issue for German: #1054

Hermann12, 2020-01-10:
Updated and tested the .mo and .po. Looks good now in German. Please review my comments, if possible. I have also realized, that the item under Tools  | "Manage packages ..." and "Manage Plug-ins" are very similar. maybe this can be handeled by a filer or tab in the same Pop Up Window? Only a idea.

How do you get the translated file? I can't upload the files on github folder.

closed 2020-10-28 by aivarannamaa